wongsyrone/openwrt-1 #19: trunk 版本编译tp-wr2041nv1后wifi无法配置

emenwin, 2015-03-03:
非常感谢详细的tp-wr2041n编译配置,但是有下面一个问题

刷机后安装 luci, 并通过web配置wifi 接口,返回错误如下:

/usr/lib/lua/luci/dispatcher.lua:433: Failed to execute arcombine dispatcher target for entry '/admin/network/wireless/radio0.network1'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/admin_network/wifi.lua:177: attempt to index field 'hwmodelist' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167>

dmesg  存在如下信息
    
[   11.890000] ath9k ar934x_wmac: failed to initialize device
[   11.900000] ath9k: probe of ar934x_wmac failed with error -5

wifi detect 输出为空 

有什么建议吗?

wongsyrone, 2015-03-03:
你给的 dmesg 信息我查过，有大神说2041nv1的无线是pci网卡，所以会出现无法初始化设备，至于目前的2041的mach设备c文件还没有特别完善，GPIO还不是特别确定

luci我也曾经编译进固件过，不过配置无线的时候并没有发现问题，因为新编译的东西直接opkg的时候会因为太新导致opkg出问题，建议将luci编译进固件

如果对mach的c文件有什么建议继续讨论吧

wongsyrone, 2015-03-05:
@emenwin  你编译的时候打开 V=99 选项了吗，这周末我试着编译一下看看

wongsyrone, 2015-03-11:
Closed due to inactivity.

wongsyrone, 2015-03-13:
Bug confirmed, missing Wi-Fi 
can not initialize PCI

wongsyrone, 2015-03-19:
https://github.com/wongsyrone/openwrt-1/blob/master/target/linux/ar71xx/generic/profiles/tp-link.mk#L381
need `kmod-ath9k` ?

wongsyrone, 2015-04-11:
可能是ART的偏移问题？还没测

wongsyrone, 2015-04-18:
wr2041n是否有DSA Switch？查ttl输出
Atheros AR71xx DSA switch  DEV_DSA_H  941n.c
DSA貌似控制port name

commit 9ca03223 (2015-04-18): ar71xx: tl-wr2041n-v1: fix ART offset

fix #19
tested with luci
closed 2015-04-18 by wongsyrone