wongsyrone/openwrt-1 #44: wrt1900ac works when building with gcc 4.8

wongsyrone, 2015-06-12:
official gcc 4.9 and self-added 4.9 -------- kernel boot stuck

MIPS can use gcc 4.9 and works well

wongsyrone, 2015-06-14:
someone says DTB not attached

wongsyrone, 2015-06-14:
based on link: https://developer.ridgerun.com/wiki/index.php/Common_Problems_-_Linux_kernel_doesnt_boot
we need kernel console configuration, machine ID and well-built kernel image to boot, and kernel console information was coded in DTS file.

look for:
[ ] compile DTB correctly
[ ] compile kernel and make image correctly

wongsyrone, 2015-06-14:
https://github.com/wongsyrone/openwrt-1/blob/master/target/linux/mvebu/patches-3.18/001-add_mamba_support.patch#L101     console info here, but this info not passed to linux kernel, maybe DTB corrupted or kernel image corrupted

wongsyrone, 2015-06-17:
DT warning 
https://lists.ozlabs.org/pipermail/linuxppc-dev/2009-February/068170.html

wongsyrone, 2015-06-17:
try
http://lists.infradead.org/pipermail/linux-arm-kernel/2014-January/226851.html

wongsyrone, 2015-06-18:
try to add buildroot's gcc 4.9.x patches

wongsyrone, 2015-06-18:
check openembedded patches
https://github.com/openembedded/oe-core/tree/master/meta/recipes-devtools/gcc/gcc-4.9

wongsyrone, 2015-06-19:
Reported: https://dev.openwrt.org/ticket/19916

wongsyrone, 2015-06-19:
Giving up, and waiting upstream fix

wongsyrone, 2015-06-19:
bootargs = "console=ttyS0,115200";

wongsyrone, 2015-06-21:
finally https://github.com/openwrt-mirror/openwrt/commit/53a77a939215c571f27c44c436cf5f715ffc6235 , waiting......

wongsyrone, 2015-07-14:
switching to self-added gcc 5.1 and need test

wongsyrone, 2015-07-22:
failed on gcc 4.9, 5.1 and 5.2

wongsyrone, 2015-08-17:
Still broken on official GCC 5.2.0

plntyk, 2015-08-18:
can you try to insert / add to KCFLAGS (i dont know where - probably somewhere in kernel make templates)
```
-fno-ipa-sra
```
source: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=65932

wongsyrone, 2015-08-19:
Thanks, I will try.

closed 2015-08-19 by wongsyrone

wongsyrone, 2015-08-19:
@plntyk 
Please refer to https://github.com/wongsyrone/openwrt-1/commit/35308f2435ec75192324d68817e7702e3b257e71 and https://github.com/wongsyrone/openwrt-1/commit/93c0a5173414cfa7684547de3c3a1f3dc4240383
If you want to send these changes to OpenWrt's mailing list, feel free to do it.

wongsyrone, 2015-09-03:
nbd pushed a better solution from gcc guys, I will revert my changes.

commit 72a58645 (2015-08-19): toolchain/gcc: finally ARM/mvebu's kernel works.

I can safely enable GCC 5.2.0 for it.
commits: 35308f2435ec75192324d68817e7702e3b257e71 and 93c0a5173414cfa7684547de3c3a1f3dc4240383
Thanks @plntyk

close #44