wongsyrone/openwrt-1 #63: fdisk编译不过，请问何解？

kuoruan, 2015-08-26:
Package fdisk is missing dependencies for the following libraries:
libfdisk.so.1

wongsyrone, 2015-08-26:
目前代码有些问题，不过暂时没时间弄

kuoruan, 2015-08-26:
能告诉一下fdisk包的目录么？找了一下没找到，就找到util-linux下Makefile里边定义了一下

wongsyrone, 2015-08-26:
就是util-linux的一个组件，一般不用单独安装fdisk

kuoruan, 2015-08-26:
```code
make[4]: Entering directory '/home/liao/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.10/vsftpd-notls/vsftpd-3.0.3'
mipsel-openwrt-linux-musl-gcc -o vsftpd main.o utility.o prelogin.o ftpcmdio.o postlogin.o privsock.o tunables.o ftpdataio.o secbuf.o ls.o postprivparent.o logging.o str.o netstr.o sysstr.o strlist.o banner.o filestr.o parseconf.o secutil.o ascii.o oneprocess.o twoprocess.o privops.o standalone.o hash.o tcpwrap.o ipaddrparse.o access.o features.o readwrite.o opts.o ssl.o sslslave.o ptracesandbox.o ftppolicy.o sysutil.o sysdeputil.o seccompsandbox.o  -L/home/liao/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.10/usr/lib -L/home/liao/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.10/lib -L/home/liao/openwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/usr/lib -L/home/liao/openwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/lib -znow -zrelro -lcrypt -lnsl
/home/liao/openwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/lib/gcc/mipsel-openwrt-linux-musl/4.8.3/../../../../mipsel-openwrt-linux-musl/bin/ld: cannot find -lnsl
collect2: error: ld returned 1 exit status
Makefile:31: recipe for target 'vsftpd' failed
```
新问题。。。vsftpd也不过了，提示如上，找不到libnsl
我的系统是Ubuntu 14.10 64位。。

wongsyrone, 2015-08-26:
使用tls或pam版本

kuoruan, 2015-08-26:
pam版本用root账户连接不上，我找了好久都不知道怎么回事

kuoruan, 2015-08-26:
大神，能说一下vsftpd-pam怎么启用root账户登录么，root连不上啊。匿名账户可以。

wongsyrone, 2015-08-26:
安全问题root默认禁用，应该使用专用账户

wongsyrone, 2015-10-11:
@kuoruan 不好意思，fdisk的问题已经确认，正在修复

wongsyrone, 2015-10-11:
@kuoruan 问题已经修复，尝试编译最新代码

commit e5fb80ae (2015-10-11): util-linux: add libfdisk

fix #63
closed 2015-10-11 by wongsyrone