wongsyrone/openwrt-1 #64: luci-app-youku和fdisk

kuoruan, 2015-09-01:
```
http://git.oschina.net/chtomato/luci-app-youku
```
想编译个路由宝挖矿，放进去之后不能选择，用官方的源码正常。。
fdisk不能编译的问题。请问能解决么？
我想使用这个脚本，希望能解决一下。
```
http://www.godiy.net/viewthread.php?tid=14441
```
我也知道我问题很多，但是。。换过去换过来，最后我还是选择了你的源码，我真的很喜欢你源码编译出来的固件，希望你能帮忙解决一下。
额，还有：
```
https://github.com/bywayboy/openwrt-packages
```
这个subversion希望你能整合进去，我试过了，编译正常，使用应该也是正常的。
希望你能帮助解决一下以上的问题。。非常感谢。。

wongsyrone, 2015-09-01:
fdisk需要在busybox里面作为applet存在，而不是通常意义的软件包，这个应该修改默认config就行，fdisk我会留意测试，实际上使用的ext系列的文件系统在任何linux下都可做分区格式化的操作，而且OP下不会考虑添加NTFS这类文件系统的支持，包括自动挂载，linux内核不是为它而生的。
subversion为什么不用官方packages repo里面的包呢？
至于优酷，如果里面使用了预编译的二进制，则不会考虑添加，晚上回去我看一下。

kuoruan, 2015-09-01:
主要官方包里边的subversion是1.6版的，版本太低。
我把全部清空了，正在测试完全重新编译，试试fdisk这次能不能编译上。
另外，您这个源码编译luci-app-qosv4貌似不能正常限速，我也在重新测试。

wongsyrone, 2015-09-01:
我这里并没有qosv4

wongsyrone, 2015-09-01:
@kuoruan 
busybox默认开启fdisk的修改：
开启 https://github.com/wongsyrone/openwrt-1/blob/master/package/utils/busybox/Config-defaults.in#L1387-L1395， 将默认的 n 修改成 y，https://github.com/wongsyrone/openwrt-1/blob/master/package/utils/busybox/Config-defaults.in#L1411 这个也可以改成y，开启高级特性

wongsyrone, 2015-09-01:
luci-app-youku /  files /  usr /  bin中有预编译的二进制，所以不予考虑。

kuoruan, 2015-09-01:
感谢，另外，请问能帮忙看看luci-app-qosv4为什么不能正常限速么？
```
https://github.com/maz-1/openwrt-maz1
```
上面那个是别人做的，
```
https://github.com/kuoruan/luci-app-qosv4
```
这个是我收集的，非常感谢。。

wongsyrone, 2015-09-01:
这个你至少要提供运行的日志，否则不好确定问题

kuoruan, 2015-09-01:
也就是说，优酷那个没办法加进去了么？

wongsyrone, 2015-09-01:
是的

kuoruan, 2015-09-02:
fdisk还是无法编译
```
mkdir -p /home/liao/openwrt/bin/ramips/packages /home/liao/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.10/util-linux-2.26.2/ipkg-ramips_24kec/fdisk/CONTROL /home/liao/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.10/pkginfo
install -d -m0755 /home/liao/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.10/util-linux-2.26.2/ipkg-ramips_24kec/fdisk/usr/sbin
install -m0755 /home/liao/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.10/util-linux-2.26.2/.libs/fdisk /home/liao/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.10/util-linux-2.26.2/ipkg-ramips_24kec/fdisk/usr/sbin/
find /home/liao/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.10/util-linux-2.26.2/ipkg-ramips_24kec/fdisk -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package fdisk is missing dependencies for the following libraries:
libfdisk.so.1
Makefile:605: recipe for target '/home/liao/openwrt/bin/ramips/packages/base/fdisk_2.26.2-1_ramips_24kec.ipk' failed
make[3]: *** [/home/liao/openwrt/bin/ramips/packages/base/fdisk_2.26.2-1_ramips_24kec.ipk] Error 1
make[3]: Leaving directory '/home/liao/openwrt/package/utils/util-linux'
package/Makefile:191: recipe for target 'package/utils/util-linux/compile' failed
make[2]: *** [package/utils/util-linux/compile] Error 2
make[2]: Leaving directory '/home/liao/openwrt'
package/Makefile:188: recipe for target '/home/liao/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.10/stamp/.package_compile' failed
make[1]: *** [/home/liao/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.10/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/liao/openwrt'
/home/liao/openwrt/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 2
```

wongsyrone, 2015-09-02:
我晚上试一下在arm/mvebu下编译

wongsyrone, 2015-09-02:
busybox的fdisk没问题，已测试

closed 2015-09-02 by wongsyrone

kuoruan, 2015-09-02:
楼主，fdisk还是编译不过，你说的那几个地方全改了，我还make dirclean了，tmp也删了，编译的时候还是报相同的错误。
我实在不知道是哪里的问题了。。

wongsyrone, 2015-09-02:
看看有没有单独的fdisk软件包，然后取消掉

wongsyrone, 2015-09-02:
开启详细log方法：
打开make menuconfig里面的Advanced configuration options (for developers)，接着开启Enable log files during build process，这样在使用make V=99的时候自动会生成logs文件夹，里面详细记录了各个阶段的日志文件

kuoruan, 2015-09-02:
也就是openwrt里边的那个fdisk包不能选中么？选中就会报错？

wongsyrone, 2015-09-03:
是

kuoruan, 2015-09-04:
额，我又来了，你知道有哪个工具可以实现单IP限速么？石像鬼感觉太高级了，我连续尝试添加编译了luci-app-qos-emong和luci-app-qosv4都发现无法正常使用里边的单IP限速，我看了你做的这个源码，imq的补丁都全部打好了的。

wongsyrone, 2015-09-04:
单IP限速貌似只有tomato，不过貌似tomato的也是通过imq或者其他进行限速的，你可以研究一下命令。