wongsyrone/openwrt-1 #73: explore samba 4.3.x

wongsyrone, 2015-10-14:


wongsyrone, 2015-10-17:
https://bugzilla.samba.org/show_bug.cgi?id=10128

> I built samba 4.2.0 on a custom embedded system now, and the major hurdles to overcome were as follows:
> libldap.so depends on other shared libraries (openssl). With a native linker, the linker would typically find those, but with a cross linker, the -L parameter is not enough (it is being used for finding -l<lib> but not for dependencies of those libs). On samba3 there was the LDAP_LIBS configuration variable to handle this. Here came -rpath-link switch to the rescue - for each -L switch in LDFLAGS I also added a corresponding -rpath-link switch, telling the linker where to look for those dependent libs (e.g. if LDFLAGS was "-L /foo" now it's "-L /foo -Xlinker -rpath-link=/foo)
> 
> Python (sigh). You really need a functioning python-config for the target, that can run on the host. So that when you run "python-config --includes" it will print out the include directories for the target's python, on the build machine. Python 2.7 doesn't have that out-of-the box. You can backport the fix from Python 3, as buildroot are doing, but I had to tweak that too. Bottom line is that python-config --includes must be correct. Then you put it in configure's environment (PYTHON_CONFIG=/path/to/target/python-config ./configure)
> 
> Strangely, for Python libraries and link flags, python-config is not consulted. I wish it were because if you get it right, that solves everything. There's a partial solution in the form of python_LIBDIR (which allows you ONLY to add to linker search path) and python_LDFLAGS (which allows you ONLY to set more link flags). Best thing is to set them to empty string, relying on existing LDFLAGS that already points to python libs. However, the additional libraries to link against (additional -l flags) are taken from the build machine python (distutils) and that's a bug. Luckily for me they were the same, but even if my target python was statically-linked, I would run into a problem b/c the order of libraries included is dependencies first, python second. 
> 
> Another issue I encountered was that my embedded kernel didn't support IPv6 and I couldn't find a way in the build system to disable IPv6 - IPv6 was enabled based on it being supported by libc. That means smbd cannot listen to connections. My fix for that was simply to add IPv6 support.
> 
> So to summarize:
> PYTHON_CONFIG=/path/to/my/target python-config \
> python_LDFLAGS="" \
> python_LIBDIR="" \
> LDFLAGS="${LDFLAGS} -Xlinker -rpath-link=/path/to/openssl/libs \
> ./configure --cross-compile --cross-answeres=/my/answers/file ....

wongsyrone, 2015-10-17:
BUILDROOT PYTHON PATCHES
001-remove-host-header-path.patch
009-no-termcap-host-path.patch
002-fix-get-python-inc.patch
004-sysconfigdata-install-location.patch
005-pyc-pyo-conditional.patch
006-cross-compile-getaddrinfo.patch
007-disable-extensions.patch
011-remove-python-symlink.patch
014-abort-on-failed-modules.patch
016-serial-ioctl-workaround.patch
018-fix-add-gcc-paths-logic.patch
114-remove-idle-editor.patch

--------------------------------------------------------

OPENWRT PYTHON PATCHES
002-do-not-add-include-dirs-when-cross-compiling.patch
003-do-not-compile-tests-at-build.patch
004-do-not-write-bytes-codes.patch
005-fix-libffi-x86-64-configure.patch
009-do-not-use-dblib_dir-when-cross-compiling.patch
010-do-not-add-rt-lib-dirs-when-cross-compiling.patch
011-do-not-prefer-ncursesw.patch

--------------------------------------------------------

IMPORTANT FOR SAMBA
010-fix-python-config.patch

wongsyrone, 2015-10-18:
Missing dependencies:
Package samba43-server is missing dependencies for the following libraries:
libCHARSET3-samba4.so
libauth-samba4.so
libcli-cldap-samba4.so
libcli-nbt-samba4.so
libcli-smb-common-samba4.so
libcliauth-samba4.so
libdcerpc-samba-samba4.so
liberrors-samba4.so
libgenrand-samba4.so
libgse-samba4.so
liblibcli-lsa3-samba4.so
liblibsmb-samba4.so
libmsrpc3-samba4.so
libndr-nbt.so.0
libndr-samba-samba4.so
libndr-standard.so.0
libndr.so.0
libnetapi.so.0
libpopt-samba3-samba4.so
libreplace-samba4.so
libsamba-cluster-support-samba4.so
libsamba-debug-samba4.so
libsamba-hostconfig.so.0
libsamba-passdb.so.0
libsamba-security-samba4.so
libsamba-sockets-samba4.so
libsamba-util.so.0
libsamba3-util-samba4.so
libsecrets3-samba4.so
libserver-role-samba4.so
libsmbconf.so.0
libsmbd-base-samba4.so
libsmbd-shim-samba4.so
libsmbregistry-samba4.so
libsocket-blocking-samba4.so
libsys-rw-samba4.so
libtalloc.so.2
libtdb.so.1
libtevent.so.0
libutil-cmdline-samba4.so
libutil-tdb-samba4.so

wongsyrone, 2015-10-18:
Solution 1: just copy these so files to /usr/lib

wongsyrone, 2015-10-18:
done in ad1327a8213def6d0135a47ee2e051efef24636b

wongsyrone, 2015-10-18:
binary too BIG

pentie, 2015-10-23:
@wongsyrone 
I have a better on the patching the python-config, I explored the source, python-config reads some environment var. to add up the output path. Thus all we need is set up such env vars.

add this lines in Makefile before running configure:
```
export _python_sysroot=$(STAGING_DIR)
export _python_prefix=/usr
export _python_exec_prefix=/usr/bin
```

A normal version of python-config will output the correct path of include in the staging_dir

wongsyrone, 2015-10-23:
@pentie 
I appreciate your work but waf needs cflags, ldflags, etc. Have you tested with your changes?
We'd better focus on reducing package size IMO.

pentie, 2015-10-27:
@wongsyrone  python2-config will still provide cflags/ldflags correctly, this env variables only amend to the path of the -L -I flags. The python part is flawless.

I got the package compiles, but it fails on the final linking the nmbd(ld segfaults, nothing to do with package), I think there is something wrong with the offcial SDK, I'm going to compile the build tree from scratch.

wongsyrone, 2015-10-27:
@pentie 
You can try this tree, it works.

pentie, 2015-10-27:
for the record I encounter for now, I did this changes to the Makefile
```diff
--- /tmp/samba43related/samba43/Makefile        2015-10-22 11:43:35.000000000 +0800
+++ Makefile    2015-10-23 14:14:51.760517000 +0800
@@ -61,6 +61,10 @@
 
 TARGET_LDFLAGS += -Wl,--gc-sections
 
+export _python_sysroot=$(STAGING_DIR)
+export _python_prefix=/usr
+export _python_exec_prefix=/usr/bin
+
 # Use host compiled python2 as interpreter
 # Use target python2-config to retrive cflags, ldflags, libs, etc.
 # Must have python2 patched to make python2-config works
@@ -136,7 +140,7 @@
                ./buildtools/bin/waf configure -j$(shell nproc) \
                        --cross-compile \
                        --cross-answers=$(PKG_BUILD_DIR)/cache.txt \
-                       --hostcc=$(HOSTCC) \
+                       --hostcc="$(HOSTCC)" \
                        $(SAMBA_CONFIGURE_ARGS) \
                        --with-lockdir=/var/lock \
                        --with-logfilebase=/var/log \
```

pentie, 2015-10-27:
`$(HOSTCC) ` here is "ccache gcc", without quotes, waf reads gcc as a command, breaks everything ( took my quite a while to find out, ah)

mormts, 2015-10-28:
@wongsyrone 
[image]

collect2: ld terminated with signal 11 [Segmentation fault]

How do you handle this change?

wongsyrone, 2015-10-28:
@mormts 
I'm using trunk code and musl-libc, haven't noticed this kind of error

wongsyrone, 2015-10-28:
@mormts  
It seems yours is similar to @pentie 's, pls see his comment above.

mormts, 2015-10-28:
[image]

Use the trunk musl return is also such a problem

wongsyrone, 2015-10-28:
@mormts  SDK or toolchain built by yourself?

mormts, 2015-10-28:
@wongsyrone 
no

wongsyrone, 2015-10-28:
my configuration:
GCC: 5.2 + SSP enabled + -D_FORTIFY_SOURCE=1 + Full RELRO
gdb: 7.8
binutils: 2.25.1
target: mvebu (ARM)

I have no idea why you guys have such a problem.

wongsyrone, 2015-10-28:
@pentie @mormts 
pls remove this line: https://github.com/wongsyrone/openwrt-1/blob/master/package/external/samba43related/samba43/Makefile#L154 and re-compile.

mormts, 2015-10-28:
Ok good I try

wongsyrone, 2015-10-28:
@pentie @mormts 
Also, pls discard all changes made by yourselves if not work, and use my package-1 repo instead.

mormts, 2015-10-28:
@wongsyrone 
openwrt-1 packages-1
The same is the error
Memory size Have a relationship?
1G

wongsyrone, 2015-11-14:
https://archives.gentoo.org/gentoo-dev/message/d75c3ba798825052daee35e846e79797
https://lists.samba.org/archive/samba/2015-September/194453.html

  Some important options:

   --with-static-modules=STATIC_MODULES
            Comma-separated list of names of modules to statically link in
   --with-shared-modules=SHARED_MODULES
            Comma-separated list of names of modules to build shared
    --bundled-libraries=BUNDLED_LIBS
            comma separated list of bundled libraries. May include !LIBNAME to disable bundling a library. Can be 'NONE' or 'ALL' [auto]
    --private-libraries=PRIVATE_LIBS
            comma separated list of normally public libraries to build instead as private libraries. May include !LIBNAME to disable making a library private. Can be 'NONE' or 'ALL' [auto]
    --builtin-libraries=BUILTIN_LIBRARIES
            command separated list of libraries to build directly into binaries [NONE]
    --nonshared-binary=NONSHARED_BINARIES
            Disable use of shared libs for the listed binaries

wongsyrone, 2015-11-14:
disable libraries by appending `enabled=False` to SAMBA(3)_LIBRARY definations

library list:

grep SAMBA_LIBRARY -r -l
libcli/util/wscript_build
libcli/registry/wscript_build
libcli/security/wscript_build
libcli/auth/wscript_build
libcli/cldap/wscript_build
libcli/ldap/wscript_build
libcli/nbt/wscript_build
libcli/named_pipe_auth/wscript_build
libcli/smb/wscript
librpc/wscript_build
buildtools/wafsamba/wafsamba.py
buildtools/wafsamba/samba_python.py
buildtools/wafsamba/samba3.py
source3/wscript_build
libds/common/wscript_build
auth/credentials/wscript_build
auth/wscript_build
auth/gensec/wscript_build
python/wscript_build
nsswitch/libwbclient/wscript
nsswitch/wscript_build
third_party/zlib/wscript
third_party/popt/wscript
lib/ldb/wscript
lib/util/wscript_build
lib/tdb/wscript
lib/torture/wscript_build
lib/tevent/wscript
lib/uid_wrapper/wscript
lib/param/wscript_build
lib/socket/wscript_build
lib/socket_wrapper/wscript
lib/wscript_build
lib/addns/wscript_build
lib/resolv_wrapper/wscript
lib/talloc/wscript
lib/krb5_wrap/wscript_build
lib/ldb-samba/wscript_build
lib/nss_wrapper/wscript
lib/dbwrap/wscript_build
lib/tdb_wrap/wscript_build
lib/replace/wscript
dfs_server/wscript_build
source4/libcli/wbclient/wscript_build
source4/libcli/wscript_build
source4/libcli/ldap/wscript_build
source4/rpc_server/wscript_build
source4/librpc/wscript_build
source4/param/wscript_build
source4/auth/ntlm/wscript_build
source4/auth/wscript_build
source4/auth/kerberos/wscript_build
source4/smbd/wscript_build
source4/dns_server/wscript_build
source4/libnet/wscript_build
source4/ntvfs/posix/wscript_build
source4/ntvfs/wscript_build
source4/lib/policy/wscript_build
source4/lib/registry/wscript_build
source4/lib/socket/wscript_build
source4/lib/http/wscript_build
source4/lib/events/wscript_build
source4/lib/samba3/wscript_build
source4/lib/messaging/wscript_build
source4/lib/cmdline/wscript_build
source4/cluster/wscript_build
source4/dsdb/samdb/ldb_modules/wscript_build
source4/dsdb/wscript_build
source4/kdc/wscript_build

grep SAMBA3_LIBRARY -r -l
buildtools/wafsamba/samba3.py
source3/libsmb/wscript
source3/auth/wscript_build
source3/wscript_build
source3/pam_smbpass/wscript_build
source3/modules/wscript_build
source3/winbindd/wscript_build
nsswitch/wscript_build

wongsyrone, 2016-03-03:
It seems we can't disable krb5 in samba 4.3.x, which may increase ipk size. Extroot or high-end devices should eliminate the package size concern.

I'm closing this. To be honest, I don't know how to reduce it.

closed 2016-03-03 by wongsyrone

hbl0307106015, 2016-05-03:
please help me for the below compilation error:

[ 224/1998] Generating param_functions.c
[ 225/1998] Generating param_functions.h
[ 226/1998] Generating param_local.h
[ 227/1998] Generating param_global.h
[ 228/1998] Generating param_table_gen.c
[ 229/1998] Compiling IDL librpc/idl/atsvc.idl
[ 230/1998] Compiling IDL librpc/idl/auth.idl
[ 231/1998] Compiling IDL librpc/idl/drsuapi.idl
/home/tymon/work/openwrt/git/official/trunk/qca/build_dir/target-arm_cortex-a9+neon-vfpv4_musl-1.1.14_eabi/samba-4.4.0/librpc/idl/atsvc.idl:5:1: error: expected identifier or '(' before '[' token
 [ uuid("1ff70682-0a51-30e8-076d-740be8cee98b"),
 ^
/home/tymon/work/openwrt/git/official/trunk/qca/build_dir/target-arm_cortex-a9+neon-vfpv4_musl-1.1.14_eabi/samba-4.4.0/librpc/idl/drsuapi.idl:3:8: error: expected '=', ',', ';', 'asm' or '__attribute__' before string constant
 import "security.idl", "misc.idl", "lsa.idl", "samr.idl";
        ^
/home/tymon/work/openwrt/git/official/trunk/qca/build_dir/target-arm_cortex-a9+neon-vfpv4_musl-1.1.14_eabi/samba-4.4.0/librpc/idl/drsuapi.idl:5:1: error: expected identifier or '(' before '[' token
 [
 ^
/home/tymon/work/openwrt/git/official/trunk/qca/build_dir/target-arm_cortex-a9+neon-vfpv4_musl-1.1.14_eabi/samba-4.4.0/librpc/idl/auth.idl:12:8: error: expected '=', ',', ';', 'asm' or '__attribute__' before string constant
 import "misc.idl", "security.idl", "lsa.idl", "krb5pac.idl";
        ^
/home/tymon/work/openwrt/git/official/trunk/qca/build_dir/target-arm_cortex-a9+neon-vfpv4_musl-1.1.14_eabi/samba-4.4.0/librpc/idl/auth.idl:13:1: error: expected identifier or '(' before '[' token
 [
 ^
Failed to parse /home/tymon/work/openwrt/git/official/trunk/qca/build_dir/target-arm_cortex-a9+neon-vfpv4_musl-1.1.14_eabi/samba-4.4.0/librpc/idl/atsvc.idl at /home/tymon/work/openwrt/git/official/trunk/qca/build_dir/target-arm_cortex-a9+neon-vfpv4_musl-1.1.14_eabi/samba-4.4.0/pidl/pidl line 610.
Failed to parse /home/tymon/work/openwrt/git/official/trunk/qca/build_dir/target-arm_cortex-a9+neon-vfpv4_musl-1.1.14_eabi/samba-4.4.0/librpc/idl/auth.idl at /home/tymon/work/openwrt/git/official/trunk/qca/build_dir/target-arm_cortex-a9+neon-vfpv4_musl-1.1.14_eabi/samba-4.4.0/pidl/pidl line 610.
Failed to parse /home/tymon/work/openwrt/git/official/trunk/qca/build_dir/target-arm_cortex-a9+neon-vfpv4_musl-1.1.14_eabi/samba-4.4.0/librpc/idl/drsuapi.idl at /home/tymon/work/openwrt/git/official/trunk/qca/build_dir/target-arm_cortex-a9+neon-vfpv4_musl-1.1.14_eabi/samba-4.4.0/pidl/pidl line 610.
Waf: Leaving directory `/home/tymon/work/openwrt/git/official/trunk/qca/build_dir/target-arm_cortex-a9+neon-vfpv4_musl-1.1.14_eabi/samba-4.4.0/bin'
Build failed:
 -> task failed (err #1): 
	{task: PIDL_ATSVC atsvc.idl,CUtil.pm,Compat.pm,Dump.pm,Expr.pm,IDL.pm,NDR.pm,ODL.pm,ClientNDR.pm,ServerNDR.pm,Template.pm,Header.pm,Proxy.pm,Stub.pm,Header.pm,Client.pm,Parser.pm,Server.pm,Python.pm,TDR.pm,Template.pm,Samba4.pm,Typelist.pm,Util.pm,Conformance.pm,NDR.pm,Pidl.pm,Driver.pm -> atsvc.h,ndr_atsvc.c,ndr_atsvc.h,srv_atsvc.c,srv_atsvc.h,ndr_atsvc_s.c,ndr_atsvc_c.c,ndr_atsvc_c.h,py_atsvc.c}
 -> task failed (err #1): 
	{task: PIDL_DRSUAPI drsuapi.idl,CUtil.pm,Compat.pm,Dump.pm,Expr.pm,IDL.pm,NDR.pm,ODL.pm,ClientNDR.pm,ServerNDR.pm,Template.pm,Header.pm,Proxy.pm,Stub.pm,Header.pm,Client.pm,Parser.pm,Server.pm,Python.pm,TDR.pm,Template.pm,Samba4.pm,Typelist.pm,Util.pm,Conformance.pm,NDR.pm,Pidl.pm,Driver.pm -> drsuapi.h,ndr_drsuapi.c,ndr_drsuapi.h,srv_drsuapi.c,srv_drsuapi.h,ndr_drsuapi_s.c,ndr_drsuapi_c.c,ndr_drsuapi_c.h,py_drsuapi.c}
 -> task failed (err #1): 
	{task: PIDL_AUTH auth.idl,CUtil.pm,Compat.pm,Dump.pm,Expr.pm,IDL.pm,NDR.pm,ODL.pm,ClientNDR.pm,ServerNDR.pm,Template.pm,Header.pm,Proxy.pm,Stub.pm,Header.pm,Client.pm,Parser.pm,Server.pm,Python.pm,TDR.pm,Template.pm,Samba4.pm,Typelist.pm,Util.pm,Conformance.pm,NDR.pm,Pidl.pm,Driver.pm -> auth.h,ndr_auth.c,ndr_auth.h,srv_auth.c,srv_auth.h,ndr_auth_s.c,ndr_auth_c.c,ndr_auth_c.h,py_auth.c}
make[2]: *** [/home/tymon/work/openwrt/git/official/trunk/qca/build_dir/target-arm_cortex-a9+neon-vfpv4_musl-1.1.14_eabi/samba-4.4.0/.built] Error 1
make[2]: Leaving directory `/home/tymon/work/openwrt/git/official/trunk/qca/package/network/services/samba4'
make[1]: *** [package/network/services/samba4/compile] Error 2
make[1]: Leaving directory `/home/tymon/work/openwrt/git/official/trunk/qca'
make: *** [package/network/services/samba4/compile] Error 2

@wongsyrone