wongsyrone/openwrt-1 #76: issues with imq patch on kernel 4.1.11

shangjiyu, 2015-10-26:
it seems nf_internals.h has been updated and the 999-imq.patch in [target/linux/generic/patches-4.1/999-imq.patch](https://github.com/wongsyrone/openwrt-1/blob/master/target/linux/generic/patches-4.1/999-imq.patch) should be updated too and is this right?

```diff
--- a/net/netfilter/nf_internals.h
+++ b/net/netfilter/nf_internals.h
@@ -18,7 +18,7 @@ unsigned int nf_iterate(struct list_head

 /* nf_queue.c */
 int nf_queue(struct sk_buff *skb, struct nf_hook_ops *elem,
-            struct nf_hook_state *state, unsigned int queuenum);
+            struct nf_hook_state *state, unsigned int queuenum, unsigned int queuetype);
 void nf_queue_nf_hook_drop(struct nf_hook_ops *ops);
 int __init netfilter_queue_init(void);
```

wongsyrone, 2015-10-26:
Any output at compiling time?

shangjiyu, 2015-10-26:
building in progress after update this patch which first time comes up with an patch error, hunk error info just forgot...

wongsyrone, 2015-10-26:
I will look into this issue ASAP.

wongsyrone, 2015-10-26:
I've taken a quick look, it should be a diff context mismatch issue since the target, mvebu, doesn't have linux 4.1 support yet. You can try `make target/linux/{clean,prepare} V=99 QUILT=1` and use `quilt push -f` when quilt won't let you go.

shangjiyu, 2015-10-27:
sorry, mine was 690-imq.patch and build passed after update the patch

closed 2015-10-27 by shangjiyu