wongsyrone/openwrt-1 #85: Problems compiling Samba-4.3.x

Andy2244, 2016-02-25:
Hi,

i followed this guide [https://wiki.openwrt.org/doc/howto/buildroot.exigence](url) to get a working samba 4.3 server, but did run into some problems.

**Problem 1)** The 4.3.4 file is no longer in the ftp pub, so you have to change the makefile to either use the new 4.3.5 version or point to the https url instead of the ftp, for the 4.3.4 version [(.](https://download.samba.org/pub/samba/stable/samba-4.3.4.tar.gz)
 _(I tried both the 4.3.4 and 4.3.5 versions, with similar results.)_

I followed the basic openWRT build steps, but was not sure if the $PATH manipulation was still needed as described here? [https://wiki.openwrt.org/doc/howto/build](url)

**Here are my basic steps under Ubuntu 15.10 server:**
```
sudo apt-get update
sudo apt-get install git-core build-essential libssl-dev libncurses5-dev unzip gawk
sudo apt-get install subversion mercurial
git clone git://github.com/wongsyrone/openwrt-1.git

cd openwrt-1
./scripts/feeds update -a
./scripts/feeds install -a

make defconfig
make prereq
make menuconfig

-> Target System: (Marvel Armada 37x/38x/XP)
-> selected as Module "M": 
Network/Samba43-server + async support
Languages/Python/python-light
-> saved + exit

make
```

**Problem 2)** The changelog notes:
`# Must have python2 patched to make python2-config works`
I suspect thats whats missing, since i get this error:
```
Checking for program python2.7-config                                             : /home/andy/openwrt-1/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.12_eabi/usr/bin/python2-config
/bin/sh: 1: /home/andy/openwrt-1/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.12_eabi/usr/bin/python2-config: not found
Traceback (most recent call last):
  File "./buildtools/bin/waf", line 76, in <module>
```
For me the python2.7-config is inside:
```
openwrt-1/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.12_eabi/host/bin
not
openwrt-1/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.12_eabi/usr/bin
```
I can copy the missing python files to the /usr/bin dir, but i suspect i miss some patching/configure step?

After those changes the compile seems to succeed. So i wonder if those steps are correct or i should change something in this whole procedure? 

I just need the working samba43 server bins, which i try to ultimately run under a DD-Wrt 3.0 build on my WRT-1200AC.

thx
Andy

Andy2244, 2016-02-25:
I did run into a other problem, thats probably related to trying to run it under DD-Wrt 3.0, but maybe you know what this error means and how to fix it.

trying to use: `passdb backend = smbpasswd`
results in:
```
No builtin nor plugin backend for smbpasswd found
Failed to open passdb!
```
Any idea to get this working?

wongsyrone, 2016-02-25:
for Problem 2: use my packages repo instead,https://github.com/wongsyrone/openwrt-1/blob/master/feeds.conf.wongsyrone#L1

for the last problem: use testparam to check available options, the configurations may vary between samba versions.

wongsyrone, 2016-02-25:
You can check https://github.com/wongsyrone/openwrt-1/issues/73 for more info. BTW, this package is for testing only at this time.

Andy2244, 2016-02-26:
Oki with the fixed feeds repo the compile works without problems.
I will just switch to openWRT from DD-Wrt, since i cant figure out what the problem related to the "passdb backend" is.

Btw what is the main difference from your openWRT repo, to the master-trunk? Is it mainly the external packages or what other major changes/fixed did you have in your version?

I just wonder if i should use your repo or use the general trunk + your externals for my builds?

thx
Andy

wongsyrone, 2016-02-27:
You can clone my openwrt-1, packages-1 and luci-1 repositories and check commits for details, most of them were described in English. Please note that I will do some tests which may brick your router. You should response for your own decisions.

Andy2244, 2016-02-28:
oki i get the same error under openWRT

```
smb.conf
backend = smbpasswd tdbsam
```
-----------------------------------------
`smbpasswd -D5`
results in
```
Attempting to find a passdb backend to match tdbsam (tdbsam)
No builtin backend found, trying to load plugin
Probing module 'tdbsam'
Probing module 'tdbsam': Trying to load from /usr/lib/samba/pdb/tdbsam.so
Error loading module '/usr/lib/samba/pdb/tdbsam.so': Error loading shared library /usr/lib/samba/pdb/tdbsam.so: No such                                 file or directory
No builtin nor plugin backend for tdbsam found
Failed to open passdb!
```
So the tdb module/lib is not statically linked and also not installed dynamically?

I tried to add the the tdb modules to the "--with-static-modules=", but than i get strange missing reference errors while waf is trying to link the tdb/tools?

Some hint how to get the pdb_tdbsam, pdb_smbpasswd statically linked?

thx
Andy

wongsyrone, 2016-02-28:
How about use `smbpasswd` only as auth backend?

Andy2244, 2016-02-28:
Dosn't matter both backends rely on tdb, so the error is the same.

Do you have 4.3 running on your router or you still use the 4.0 version?
I just wonder what working backend there is atm in the 4.3 version, since ldap/ac-dc is disabled and tdb doesn't work, there should be no working backend at all atm.

I assume the inability to statically link the tdb module is related to the cross compile, but i'm no expert in this area. What i tried was to add the tdb lib that comes with openWRT, but samba 4 looks for its own newer >=1.3.7 version and wont link against the very old 1.0.6 version.

I will try to get the samba 1.3.7 version working as package dependency and see if this solves the problem for the moment. It would still be better to get the tdb that comes with samba directly linked, since tdb is the default samba4 backend atm.

bye
Andy

Andy2244, 2016-03-02:
oki closing this, i have a working version after switching from ubuntu server to a arch based system. I can't figure out what went wrong on the ubuntu system, but i have a working bin now with everything statically linked.

You still need to fix the sourceurl to include the http source, since the 4.3.4 version is not on the ftp server.

```
PKG_SOURCE_URL:= http://download.samba.org/pub/samba/stable \
	http://ftp.samba.org/pub/samba \
	http://ftp.samba.org/pub/samba/old-versions
```

I also noticed that the smbd does not create missing dirs like /var/cache/samba or /var/run/samba etc. and fails to start. Thats because of the --enable-fhs option and im not sure why this option is even used in the first place, since its disabled by default. The default used dirs seem to all point to already existing directory's inside /var/.... so this option should be either removed or the special samba dirs need to be created, since smbd wont start otherwise.

thx
Andy

PS: I did some initial tests and samba 4.3.4 runs fine so far, without any errors or warnings in the logs. So i really think you could push the samba/heimdal/luci samba4 feeds + python fix to the official repo. The size concern is the same for every big package, which is solved by exroot or having a modern router with 32MB+ nvram. You also use the samba4 naming scheme so we cans till choose to use samba36 if needed.

wongsyrone, 2016-03-02:
I enable `--enable-fhs` since buildroot does [so](https://git.busybox.net/buildroot/tree/package/samba4/samba4.mk#n98), I will dig more on the missing dirs issue. Thank you for testing this package and suggestions above. 

BTW, Can you create symlink to the multicall binary called `smbtorture` and try to execute it to see whether this module was compiled into the ipk file? If smbtorture is included, I will dig into the build system and try to disable its build scripts, which will make the ipk smaller.

In general, the multicall binary consists of smbd, nmbd, smbpasswd, smbcontrol, smbtree, smbstatus, smbcquotas, smbcacls, nmblookup and testparm. Can you test these functionalities for me?

Andy2244, 2016-03-02:
Sure can do some basic testing on the utils later today. 
Yesterday i did at least run smbd using passwd/tdb pdb backends and smbcontrol/status/testparam.

Ah btw there was a problem using passwd backend, since it seems samba could not add new users to the openwrt system. I could add a smb password to the root user and also used this combination to access the shares from my Win10 machines. I suspect thats because the adduser util is missing by default? If thats the case we should add this as a package dependency.

wongsyrone, 2016-03-02:
I usually add user to `/etc/passwd `before executing `smbpasswd -a`, you can refer to `User level access` section in the [wiki](https://wiki.openwrt.org/doc/uci/samba)

wongsyrone, 2016-03-02:
Adding user to /etc/passwd can be achieved in the Makefile like transmission does, but I think leaving this part to end user will be better hence they can custom the users themselves.

Andy2244, 2016-03-02:
Ah thats the problem than.

Btw the luci samba4 files could also use a update to support the 4.3 server, since they still refer to the samba42-server.

thx
Andy

wongsyrone, 2016-03-02:
The luci part will not be a problem since they are almost identical.

wongsyrone, 2016-03-02:
Can you post the output of smbd -b?

Andy2244, 2016-03-02:
oki 
0) my compiled samba_multicall is ~14MB
1) `smbtorture` is not compiled in
2) smbd, nmbd, smbpasswd, smbtree, smbstatus, testparm work as expected
3) smbcontrol, smbcquotas, smbcacls, nmblookup show the help without errors, but i did not really test those on a functional level
4) manually adding users to `/etc/passwd` and than using `smbpasswd -a` works
5) netbios via nmbd works and Win10 sees the shares without problems
6) read/write on a ext4 usb mounted drive works from Win10
7) `/etc/init.d/samba4 enable/start` **does not work**, manually invoking smbd -D and nmbd -D works.
```
~#/etc/init.d/samba4 enable
~# /etc/init.d/samba4 start
samba4 is disabled
```

`smbd -b`
```
Build environment:
   Built by:    andy@manjaro
   Built on:    Tue Mar  1 22:40:53 UTC 2016
   Built using: arm-openwrt-linux-muslgnueabi-gcc
   Build host:  Linux manjaro 4.4.3-1-MANJARO #1 SMP PREEMPT Fri Feb 26 23:15:14 UTC 2016 i686 GNU/Linux
   SRCDIR:      /home/andy/openwrt-1-master/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/samba-4.3.4/source3
   BUILDDIR:    /home/andy/openwrt-1-master/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/samba-4.3.4/source3

Paths:
   SBINDIR: /usr/sbin
   BINDIR: /usr/bin
   CONFIGFILE: /etc/samba/smb.conf
   LOGFILEBASE: /var/log
   LMHOSTSFILE: /etc/samba/lmhosts
   LIBDIR: /usr/lib
   MODULESDIR: /usr/lib/samba
   SHLIBEXT: so
   LOCKDIR: /var/lock
   STATEDIR: /var/lib/samba
   CACHEDIR: /var/cache/samba
   PIDDIR: /var/run
   SMB_PASSWD_FILE: /etc/samba/smbpasswd
   PRIVATE_DIR: /etc/samba

System Headers:
   HAVE_SYS_CDEFS_H
   HAVE_SYS_DIR_H
   HAVE_SYS_EPOLL_H
   HAVE_SYS_EVENTFD_H
   HAVE_SYS_FCNTL_H
   HAVE_SYS_FILE_H
   HAVE_SYS_INOTIFY_H
   HAVE_SYS_IOCTL_H
   HAVE_SYS_IPC_H
   HAVE_SYS_KERNEL_PROC_CORE_PATTERN
   HAVE_SYS_MMAN_H
   HAVE_SYS_MOUNT_H
   HAVE_SYS_PARAM_H
   HAVE_SYS_PRCTL_H
   HAVE_SYS_RESOURCE_H
   HAVE_SYS_SELECT_H
   HAVE_SYS_SENDFILE_H
   HAVE_SYS_SHM_H
   HAVE_SYS_SOCKET_H
   HAVE_SYS_STATFS_H
   HAVE_SYS_STATVFS_H
   HAVE_SYS_STAT_H
   HAVE_SYS_STROPTS_H
   HAVE_SYS_SYSCALL_H
   HAVE_SYS_SYSLOG_H
   HAVE_SYS_SYSMACROS_H
   HAVE_SYS_TERMIOS_H
   HAVE_SYS_TIMEB_H
[... 668 lines cut ...]
   vfs_unityed_media_init
   vfs_worm_init
   vfs_xattr_tdb_init

Cluster support features:
   NONE

Type sizes:
   sizeof(char):         1
   sizeof(int):          4
   sizeof(long):         4
   sizeof(long long):    8
   sizeof(uint8_t):      1
   sizeof(uint16_t):     2
   sizeof(uint32_t):     4
   sizeof(short):        2
   sizeof(void*):        4
   sizeof(size_t):       4
   sizeof(off_t):        8
   sizeof(ino_t):        8
   sizeof(dev_t):        8

Builtin modules:
   pdb_smbpasswd pdb_tdbsam pdb_wbc_sam auth_sam auth_unix auth_winbind auth_wbc auth_domain auth_builtin vfs_default nss_info_template idmap_tdb idmap_passdb idmap_nss 
```

Andy2244, 2016-03-02:
Oki also just tested a btrfs mounted partition on a ssd and the speed is really nice. I also see correct security information's on Win10, which i have never seen using samba36.

Powershell also now reports using smb Dialect 3.1.1 via `Get-SmbConnection`.

I will do some more heavy tests on the weekend, using samba 4.3.5 and a lzop compression enabled btrfs partition.

bye Andy

PS: Now i just need to get Softether VPN and XtreemFS server to also work under openWRT, for my ultimate setup :p

wongsyrone, 2016-03-03:
Can you share your smb.conf and how to get windows 10 show the samba share?

wongsyrone, 2016-03-03:
I pushed https://github.com/wongsyrone/openwrt-1/commit/dcd5f24533cf99480d8294319082b3a06ab82606, you may want to have a try.

wongsyrone, 2016-03-03:
You should change `enabled` in `/etc/config/samba4` to 1 before executing `/etc/init.d/samba4 start`. Please refer to `/etc/init.d/samba4` for more info.

Andy2244, 2016-03-03:
For me the trunk is broken, because the new git-core check is not working? Do you have the same problem on your Arch distro?

```
Build dependency: Please install Git (git-core) >= 1.7.12.2

/home/andy/openwrt-1-master/include/prereq.mk:12: recipe for target 'prereq' failed
Prerequisite check failed. Use FORCE=1 to override.
/home/andy/openwrt-1-master/include/toplevel.mk:152: recipe for target 'staging_dir/host/.prereq-build' failed
make[1]: *** [staging_dir/host/.prereq-build] Error 1
/home/andy/openwrt-1-master/include/toplevel.mk:81: recipe for target 'prepare-tmpinfo' failed
make: *** [prepare-tmpinfo] Error 2
Cannot open './tmp/.packageinfo': No such file or directory
Can't open file './tmp/.targetinfo': No such file or directory
```

wongsyrone, 2016-03-03:
You should install a new version as it says.

Andy2244, 2016-03-05:
Regarding the git error, this is/was a error in trunk and has nothing to-do with a missing version. I did a bugreport and they added some fixes today.

There are still some problems i encountered with the current version:

1) 
I have a "andy" user added via smbpasswd and is also in the passwd:
`andy:*:1000:65534:smbusers:/var:/bin/false`
The smb.conf is using:
`valid users = andy`

yet all folders/files i create are set to the root user, which than using the global masks results in the inability to access new folders/files on the share?
see:
```
	create mask = 0700
	directory mask = 0700
	force create mode = 0700
	force directory mode = 0700
```

2) I get some testparams warnings:
```
rlimit_max: increasing rlimit_max (1024) to minimum Windows limit (16384)
WARNING: lock directory /var/lock should have permissions 0755 for browsing to work
```

3) The luci "system/startup" for samba4 seems not to work at all, also the` /etc/init.d/samba4 enable/disable` commands are not actually enabling/disabling the service?
I'm under the impression that luci and the above commands should do this, instead of manually needing to edit the /etc/config/samba4 file?

4) I also would like to use the `vfs objects = btrfs` module, but it seems its not enabled/linked into the package?

5) Actually using extra options per mount like the btrfs module or "Guest only = yes" is hard, since there is no option in the luci UI to add optional parameters. So on each restart/save the /etc/samba/smb.conf gets rewritten. So it would be nice to have a optional field in the ui where we can add extra parameters per share, that simply gets added to the share section.

thx Andy

PS: Btw i was playing around with the buildroot/trunk of samab4 and i wonder how much size is saved by all the multicall changes, instead of using the default shared lib config. I did check the resulting bins/libs via readelf -d and it seems the size is similar to the multicall build ~14-16 MB?
So going forward is maintaining the multicall changes/patches worth the trouble, since we cant get even close to a 8MB compatible .img build for old routers?

wongsyrone, 2016-03-06:
I just build only with smbd, nmbd, smbstatus and testparam, but only decrease to 13.9MB. I suspect the main parts eating the space are redundant libraries, for instance, krb5, if we only need a basic file server running on the router or NAS-like device. If I add `--builtin-libraries=ALL` to the configure options, I get many undefined references. To be honest, the waf build system is not cross-compiling friendly, they should keep old and new build system simultaneously.

If size is a big concern to you, I would recommend to use samba 4.0.26 which in my repo as well, but that version is EOL to the upstream, which means will no longer get security updates and bug fixes. It is really a hard decision to make.

wongsyrone, 2016-03-06:
You can change`/etc/samba/smb.conf.template` to whatever you want, this file only contains `[global]` part, re-run `/etc/init.d/samba4 restart` and the conf file will be updated. The luci part has not been sync-ed, so use `/etc/config/samba4` atm.

Here goes an example:
```
root@OpenWrt:~# cat /etc/config/samba4 

config samba
        option enabled '1'
        option name 'OpenWrt'
        option workgroup 'WORKGROUP'
        option description 'Samba on OpenWrt'
        option charset 'UTF-8'
        option homes '1'

config sambashare
        option name 'Shares'
        option path '/mnt/sda2'
        option public 'no'
        option create_mask '0700'
        option writable 'yes'
        option printable 'no'
        option users 'nas'
```

Andy2244, 2016-03-06:
Hi and thx again for the answers, i have no compiling problems anymore and i can modify the remaining config/luci files myself. 

So i will close this issue.

thx
Andy

closed 2016-03-06 by Andy2244