wongsyrone/openwrt-1 #86: flex 2.6.0 compile error

Andy2244, 2016-02-29:
using your updated tools/flex 2.6 gets me this error:
```
make[5]: Entering directory '/home/andy/openwrt-1-master/build_dir/host/flex-2.6.0/tests'
/usr/bin/env bash ../build-aux/ylwrap bison_nr_parser.y y.tab.c bison_nr_parser.c y.tab.h `echo bison_nr_parser.c | sed -e s/cc$/hh/ -e s/cpp$/hpp/ -e s/cxx$/hxx/ -e s/c++$/h++/ -e s/c$/h/` y.output bison_nr_parser.output -- bison -y -d -p test
/home/andy/openwrt-1-master/build_dir/host/flex-2.6.0/tests/bison_nr_parser.y:57.1-12: warning: deprecated directive, use '%pure-parser' [-Wdeprecated]
 %pure_parser
 ^^^^^^^^^^^^
bison: /home/andy/openwrt-1-master/staging_dir/host/share/bison/m4sugar/m4sugar.m4: cannot open: No such file or directory
Makefile:2075: recipe for target 'bison_nr_parser.c' failed
make[5]: *** [bison_nr_parser.c] Error 1
make[5]: Leaving directory '/home/andy/openwrt-1-master/build_dir/host/flex-2.6.0/tests'
Makefile:530: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
```
reverting back to 2.5.3 from trunk works

wongsyrone, 2016-02-29:
working on my Archlinux box, what's your host gcc version?

Andy2244, 2016-02-29:
gcc version 5.2.1 on 15.10 Ubuntu

wongsyrone, 2016-02-29:
this warning won't stop make on my box, I pushed a fix to disable doc and tests.

closed 2016-02-29 by Andy2244