wongsyrone/openwrt-1 #88: Compiling failed

NemoAlex, 2016-03-26:
I'm compiling with feeds.conf.wongsyrone and cfg/linksys-WRT1900AC.
Here comes error:

```
make[3]: Entering directory `/home/nemo/openwrt-1/package/utils/ubi-utils'
. /home/nemo/openwrt-1/include/shell.sh; gzip -dc /home/nemo/openwrt-1/dl/ubi-utils-1.5.2.tar.gz | tar -C /home/nemo/openwrt-1/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/ubi-utils-1.5.2/.. -xf -

Applying ./patches/010-fix-rpmatch.patch using plaintext:
patching file include/common.h
Hunk #1 succeeded at 152 (offset 15 lines).

Applying ./patches/100-fix_includes.patch using plaintext:
patching file lib/libfec.c

Applying ./patches/130-lzma_jffs2.patch using plaintext:
patching file Makefile
Hunk #1 succeeded at 3 with fuzz 2.
Hunk #2 FAILED at 84.
1 out of 2 hunks FAILED -- saving rejects to file Makefile.rej
can't find file to patch at input line 23
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/compr.c
|+++ b/compr.c
--------------------------
No file to patch.  Skipping patch.
2 out of 2 hunks ignored
can't find file to patch at input line 44
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/compr.h
|+++ b/compr.h
--------------------------
No file to patch.  Skipping patch.
2 out of 2 hunks ignored
patching file compr_lzma.c
patching file include/linux/jffs2.h
patching file include/linux/lzma.h
patching file include/linux/lzma/LzFind.h
patching file include/linux/lzma/LzHash.h
patching file include/linux/lzma/LzmaDec.h
patching file include/linux/lzma/LzmaEnc.h
patching file include/linux/lzma/Types.h
patching file lzma/LzFind.c
patching file lzma/LzmaDec.c
patching file lzma/LzmaEnc.c
can't find file to patch at input line 5014
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/mkfs.jffs2.c
|+++ b/mkfs.jffs2.c
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch failed!  Please fix ./patches/130-lzma_jffs2.patch!
make[3]: *** [/home/nemo/openwrt-1/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/ubi-utils-1.5.2/.prepared_5a16cb810844b8b234604decce1a21ca] Error 1
make[3]: Leaving directory `/home/nemo/openwrt-1/package/utils/ubi-utils'
make[2]: *** [package/utils/ubi-utils/compile] Error 2
make[2]: Leaving directory `/home/nemo/openwrt-1'
make[1]: *** [/home/nemo/openwrt-1/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/nemo/openwrt-1'
make: *** [world] Error 2
```

wongsyrone, 2016-03-26:
try `make dirclean` and recompiling

wongsyrone, 2016-03-26:
Sorry. Fixed via https://github.com/wongsyrone/openwrt-1/commit/ba81d0458e7a7580e670d50d7efedb113a620912

closed 2016-03-26 by wongsyrone

NemoAlex, 2016-03-26:
Recompiling meets new error:
```
configure: creating ./config.status
config.status: creating Makefile
config.status: creating transmission-gtk.spec
config.status: creating cli/Makefile
config.status: creating daemon/Makefile
config.status: creating extras/Makefile
config.status: creating libtransmission/Makefile
config.status: creating utils/Makefile
config.status: creating third-party/Makefile
config.status: creating third-party/dht/Makefile
config.status: creating third-party/libb64/Makefile
config.status: creating third-party/libutp/Makefile
config.status: creating third-party/libnatpmp/Makefile
config.status: creating third-party/miniupnp/Makefile
config.status: creating macosx/Makefile
config.status: creating gtk/Makefile
config.status: creating gtk/icons/Makefile
config.status: creating qt/config.pri
config.status: creating web/Makefile
config.status: creating web/images/Makefile
config.status: creating web/style/Makefile
config.status: creating web/style/jqueryui/Makefile
config.status: creating web/style/jqueryui/images/Makefile
config.status: creating web/style/transmission/Makefile
config.status: creating web/style/transmission/images/Makefile
config.status: creating web/style/transmission/images/buttons/Makefile
config.status: creating web/javascript/Makefile
config.status: creating web/javascript/jquery/Makefile
config.status: error: cannot find input file: `po/Makefile.in.in'
make[3]: *** [/home/nemo/openwrt-1/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/transmission-svn-2.92/.configured_yyyyyyyy] Error 1
make[3]: Leaving directory `/home/nemo/openwrt-1/package/external/transmission-svn'
make[2]: *** [package/external/transmission-svn/compile] Error 2
make[2]: Leaving directory `/home/nemo/openwrt-1'
make[1]: *** [/home/nemo/openwrt-1/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/nemo/openwrt-1'
make: *** [world] Error 2
```

wongsyrone, 2016-03-26:
Please pay attention to `Copying file po/Makefile.in.in` and check your full log again. It works on my box.
```
└─(06:20:%)── make package/transmission-svn/{clean,prepare} V=99                                                                                                                                ──(日,3月27)─┘
make[1]: Entering directory '/home/wong/github/openwrt-1'
make[2]: Entering directory '/home/wong/github/openwrt-1/package/external/transmission-svn'
rm -f /home/wong/github/openwrt-1/bin/mvebu/packages/base/transmission-svn-daemon_*
rm -f /home/wong/github/openwrt-1/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/stamp/.transmission-svn_installed
rm -f /home/wong/github/openwrt-1/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/packages/transmission-svn.list /home/wong/github/openwrt-1/staging_dir/host/packages/transmission-svn.list
rm -rf /home/wong/github/openwrt-1/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/transmission-svn-2.92
make[2]: Leaving directory '/home/wong/github/openwrt-1/package/external/transmission-svn'
make[1]: Leaving directory '/home/wong/github/openwrt-1'
make[1]: Entering directory '/home/wong/github/openwrt-1'
make[2]: Entering directory '/home/wong/github/openwrt-1/package/external/transmission-svn'
echo "14720" > /home/wong/github/openwrt-1/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/transmission-svn-2.92/REVISION
. /home/wong/github/openwrt-1/include/shell.sh; bzcat /home/wong/github/openwrt-1/dl/transmission-svn-2.92-14720-7fb8708efd516e629b1b33763e985189161f85ea.tar.bz2 | tar -C /home/wong/github/openwrt-1/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/transmission-svn-2.92/.. -xf - 

Applying ./patches/001-change-pt-info.patch using plaintext: 
patching file CMakeLists.txt
patching file configure.ac

Applying ./patches/002-dont-use-configure-when-autogen.patch using plaintext: 
patching file autogen.sh

Applying ./patches/003-use-internal-miniupnp.patch using plaintext: 
patching file configure.ac

Applying ./patches/030-fix-musl-build.patch using plaintext: 
patching file libtransmission/bitfield.c
patching file libtransmission/fdlimit.h

Applying ./patches/050-_GNU_SOURCE-redefined.patch using plaintext: 
patching file libtransmission/utils.c
patching file third-party/dht/dht.c
(cd /home/wong/github/openwrt-1/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/transmission-svn-2.92; ./autogen.sh )
creating libtransmission/version.h
libtoolize: putting auxiliary files in `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: copying file `m4/libtool.m4'
libtoolize: copying file `m4/ltoptions.m4'
libtoolize: copying file `m4/ltsugar.m4'
libtoolize: copying file `m4/ltversion.m4'
libtoolize: copying file `m4/lt~obsolete.m4'
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at /home/wong/github/openwrt-1/staging_dir/host/bin/automake line 3938.
configure.ac:20: installing './compile'
configure.ac:20: installing './config.guess'
configure.ac:20: installing './config.sub'
configure.ac:19: installing './install-sh'
configure.ac:19: installing './missing'
Makefile.am: installing './INSTALL'
cli/Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'
Creating aclocal.m4 ...
Running glib-gettextize...  Ignore non-fatal messages.
Copying file po/Makefile.in.in

Please add the files
  codeset.m4 gettext.m4 glibc21.m4 iconv.m4 isc-posix.m4 lcmessage.m4
  progtest.m4
from the /usr/share/aclocal directory to your autoconf macro directory
or directly to your aclocal.m4 file.
You will also need config.guess and config.sub, which you can get from
ftp://ftp.gnu.org/pub/gnu/config/.

Making aclocal.m4 writable ...
Running intltoolize...
touch /home/wong/github/openwrt-1/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/transmission-svn-2.92/.prepared_a622a605128f9f3b83da6d20cfa41aa8
make[2]: Leaving directory '/home/wong/github/openwrt-1/package/external/transmission-svn'
make[1]: Leaving directory '/home/wong/github/openwrt-1'
```
or try other Arch-based linux distributions.

NemoAlex, 2016-03-27:
Found a solution here:
https://forum.transmissionbt.com/viewtopic.php?t=12693
```
apt-get install libglib2.0-dev
```
Recompiling now.

wongsyrone, 2016-03-27:
BTW, minidlna requires gperf.

NemoAlex, 2016-03-27:
Failed again, same error.
Turns out create a empty po/Makefile.in.in will do the trick.

NemoAlex, 2016-03-27:
@wongsyrone Thanks, installing that.

wongsyrone, 2016-03-27:
try
```
glib-gettextize --version
libtoolize --help
glibtoolize --help
```
and install the corresponding package via apt.